omaraflak/Bluetooth-Library #7: Bluetooth connection issues

wdavies973, 2018-02-09:
I'm having some issues connecting to Bluetooth devices.
When I connect, the onError() callback is called with error message: ```read failed, socket might closed or timeout, read ret: -1```. I did a little bit of exploring into this issue, and it seems like creating an insecure connection would fix things (https://stackoverflow.com/a/36786031/4779937). Would you be able to add a toggle option for Secure / Insecure connect thread in your ```ConnectThread``` class?

If you'd like source code for the bug I'm getting, let me know.

wdavies973, 2018-02-09:
Also, what code triggers a device to list for incoming connections? Or is that automatically done?

JohannesDev, 2018-02-16:
Same, any updates on this?

OmarAflak, 2018-02-16:
I merged your pull request and pushed the code to bintray. You should be able to use the new version : 

`implement 'me.aflak.libraries:bluetooth:1.3.1'`

Set the second parameter to `true` to use an insecure connection.

`bluetooth.connectToDevice(device, true)`

closed 2018-02-16 by OmarAflak