omaraflak/Bluetooth-Library #10: add a discoverycallback.onDiscoveryStartedListener()

chaostools, 2018-03-09:
this maybe irrelevant or I might have not checked it thoroughly, but there is not a receiver or listener for ACTION_DISCOVERY_STARTED.
 it might not be much useful, but sometimes there are cases where one could use a listener for this. for example , if I  write bluetoothObj.startScanning() in my onCreate(), but want to handle a rotating circle animation on the starting of scanning I would prefer writing the code in the same place where i have written the code for stopping such animation,i.e my interface's object.

I would like to create a pull request for it.

(And by the way, a big thanks for creating this amazing library, it helped me understand the basic working of blue-tooth :))

OmarAflak, 2018-03-09:
Hi. Indeed I didn't add this listener. I'm currently updating the library and implementing other features requested by other developers. I'll include this one too. I will keep you updated ! :)

OmarAflak, 2018-03-09:
Hi again @chaostools. I just applied some changes to the library ! You can start using it now : 

`implementation 'me.aflak.libraries:bluetooth:1.3.2'`

closed 2018-03-09 by OmarAflak