omaraflak/Bluetooth-Library #11: function request: isScanning()

chaostools, 2018-03-10:
Hi. I really feel this  library has the potential to be used as a simple wrapper around Android's Bluetooth class(even maybe as a default in android libraries :smiley:  )
So here is a function request 
```java
    public boolean isScanning(){
               return bluetoothAdapter.isDiscovering();
    }
```
Sometimes the dev might want to restrict his app's user from pressing the "Scan" button again and again and to do so, he may want a checker function to do so.And since this is a great wrapper class around defaullt bluetooth, why not provide the dev with all the possible tools?

Thanks,
Ansh

OmarAflak, 2018-03-11:
You actually don't need this function to do that. Use `onDiscoveryStarted()` and `onDiscoveryFinished()` from `DiscoveryCallback`. When onDiscoveryStarted is called, disable the scan button; when onDiscoveryFinished is called, enable it again.

closed 2018-03-11 by OmarAflak

chaostools, 2018-03-11:
yeah i just saw i was using an older version of library.so no issues :)