omaraflak/Bluetooth-Library #14: Error:multiple broadcasts recieved for same device onStartScanning()

chaostools, 2018-03-11:
don't know if this is a flaw on my app's side or the library side, but here are the logs:
```

03-11 14:12:31.582 10699-10699/com.example.android.testing_lib E/#####################: btActions called: flags:flagBtOnOff,flagBtStartStopScan 1
03-11 14:12:32.753 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: HC-05
                                                                                        
                                                                                        :00:21:13:01:E5:6C:10
03-11 14:12:33.163 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:33.534 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:34.074 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:38.268 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:38.869 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:39.279 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:39.700 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:40.130 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:40.290 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:43.393 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:43.624 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:44.144 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:44.314 10699-10699/com.example.android.testing_lib E/#####################: DeviceFound: Micromax A104:D6:5F:E0:65:82:3B:10
03-11 14:12:47.618 10699-10699/com.example.android.testing_lib E/#####################: onFinish: scan finished
```
all i have are 2 devices, one is a cellphone and other is an arduino. But it is spawning the logs again and again for the cellphone in the discoverylistener.(i am using a slightly modified version of your library)

chaostools, 2018-03-11:
tried a few more times, its spawning the logs for everything, and not just my cell phone

chaostools, 2018-03-11:
although i have an hc 05 module, i will still look into its code. Can you tell me what ThreadHelper class is doing?

chaostools, 2018-03-11:
and yes,the current playstore's app is also listing my sungle device multiple times on start scanning

OmarAflak, 2018-03-11:
> Can you tell me what ThreadHelper class is doing?

It's simply used to run the listener in the UI thread.

> and yes,the current playstore's app is also listing my single device multiple times on start scanning .Can it be a problem of the mobile phone am using to run the app?

Can you do the test with the settings app of your phone ? Is it listing the device multiple times ?

chaostools, 2018-03-11:
yess , it seems that it is a device fault . i changed my device which has stopped multiple spawning.

closed 2018-03-11 by chaostools