omaraflak/Bluetooth-Library #16: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void java.io.OutputStream.write(byte[])' on a null object reference

badarshahzad, 2018-03-12:
Hi
 Firstly thank you for this awesome and life saving library. I am just trying to send data through bluetooth but it just give me this error.

```
03-12 17:58:56.029 22414-22414/? E/AndroidRuntime: FATAL EXCEPTION: main
                                                   Process: com.badarshahzad54.pcbs, PID: 22414
                                                   java.lang.RuntimeException: Error receiving broadcast Intent { act=sendData flg=0x10 (has extras) } in com.badarshahzad54.pcbs.service.SendAndReceive$4@d5fe5e5
                                                       at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1172)
                                                       at android.os.Handler.handleCallback(Handler.java:836)
                                                       at android.os.Handler.dispatchMessage(Handler.java:103)
                                                       at android.os.Looper.loop(Looper.java:203)
                                                       at android.app.ActivityThread.main(ActivityThread.java:6251)
                                                       at java.lang.reflect.Method.invoke(Native Method)
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063)
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924)
                                                    Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'void java.io.OutputStream.write(byte[])' on a null object reference
                                                       at me.aflak.bluetooth.Bluetooth.send(Bluetooth.java:92)
                                                       at com.badarshahzad54.pcbs.service.SendAndReceive.send(SendAndReceive.java:76)
                                                       at com.badarshahzad54.pcbs.service.SendAndReceive$4.onReceive(SendAndReceive.java:188)
                                                       at android.app.LoadedApk$ReceiverDispatcher$Args.run(LoadedApk.java:1162)
                                                       at android.os.Handler.handleCallback(Handler.java:836) 
                                                       at android.os.Handler.dispatchMessage(Handler.java:103) 
                                                       at android.os.Looper.loop(Looper.java:203) 
                                                       at android.app.ActivityThread.main(ActivityThread.java:6251) 
                                                       at java.lang.reflect.Method.invoke(Native Method) 
                                                       at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1063) 
                                                       at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:924) 

```

OmarAflak, 2018-03-12:
Hi ! Are you sure the device is connected ? Did you set the callbacks ?

closed 2018-03-13 by badarshahzad

Defaultersqwad, 2020-05-04:
Hello Sir,
I am facing same issue
i am trying to accessing send() method from another activity but facing the above issue
Maybe i am not able to set the proper Callbacks
will u please guide me ?