omaraflak/Bluetooth-Library #17: Can't create handler inside thread that has not called Looper.prepare()

arnleal09, 2018-03-12:
Hi very cool project thank you for your work, i am trying to connect to a BLE device, and this exception ei occurring
E/AndroidRuntime: FATAL EXCEPTION: Thread-4589
                  Process: com.example.aleal.app_bt_1, PID: 11989
                  java.lang.RuntimeException: Can't create handler inside thread that has not called Looper.prepare()
                      at android.os.Handler.<init>(Handler.java:200)
                      at android.os.Handler.<init>(Handler.java:114)
                      at android.widget.Toast$TN.<init>(Toast.java:655)
                      at android.widget.Toast.<init>(Toast.java:146)
                      at android.widget.Toast.makeText(Toast.java:410)
                      at com.example.aleal.app_bt_1.MainActivity$3.onConnectError(MainActivity.java:179)
                      at me.aflak.bluetooth.Bluetooth$ConnectThread$2.run(Bluetooth.java:315)
                      at me.aflak.bluetooth.ThreadHelper.run(ThreadHelper.java:15)
                      at me.aflak.bluetooth.Bluetooth$ConnectThread.run(Bluetooth.java:312)
i am using the method bluetooth.connectToName(device.getName().toString()); and 
bluetooth.connectToDevice(device);  game the same error
i am not sure if this library handle ble communications, this could be the issue? i will able to send messages after getting the connection socket open? i am testing in a Samsung j2 prime android 6.01

badarshahzad, 2018-03-13:
Hi have you implement communication callbacks? If yes did you first paired the device before connection? If yes then share the snippet code you are facing error to let us help you. Moreover, please don't forget to share your logcat error. It help us to understand quickly.

arnleal09, 2018-03-13:
Hi, thanks for your reply, yes the callbacks are implemented and they work in fact the message "read failed, socket might closed or timeout, read ret: -1" its coming from @Override public void onConnectError(BluetoothDevice device, String message), this is the code:

package com.example.aleal.app_bt_1;

import android.Manifest;
import android.bluetooth.BluetoothDevice;
import android.content.Context;
import android.content.Intent;
import android.support.v4.app.ActivityCompat;
import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.view.Gravity;
import android.view.View;
import android.widget.AdapterView;
import android.widget.ArrayAdapter;
import android.widget.ListView;
import android.widget.TextView;
import android.widget.Toast;

import java.util.ArrayList;
import java.util.List;
import java.util.UUID;

import me.aflak.bluetooth.Bluetooth;
import me.aflak.bluetooth.BluetoothCallback;
import me.aflak.bluetooth.DeviceCallback;
import me.aflak.bluetooth.DiscoveryCallback;

import static java.security.AccessController.getContext;

public class MainActivity extends AppCompatActivity {

    Bluetooth bluetooth = new Bluetooth(this);
    private ListView lv;
    ArrayList<String> devices_list = new ArrayList<>();
    String name;
    BluetoothDevice device_dongle;
    String mensaje;
    int MY_PERMISSIONS_REQUEST_ACCESS_COARSE_LOCATION=1;

    @Override
    protected void onStart() {
        super.onStart();
        bluetooth.onStart();
    }

    @Override
    protected void onStop() {
        super.onStop();
        bluetooth.onStop();
    }

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        ListView lv1 = (ListView)findViewById(R.id.element_list);
        bluetooth.setBluetoothCallback(new BluetoothCallback() {
            @Override
            public void onBluetoothTurningOn() {

            }

            @Override
            public void onBluetoothOn() {

            }

            @Override
[... 150 lines cut ...]
    public void open_drawer_2(View view) {
        // Do something in response to button click
        if (!bluetooth.isConnected())
        {
            bluetooth.connectToDevice(device_dongle);
         //   bluetooth.send("B");
        }
    }
}

this is the logcat 

03-13 08:47:54.853 20238-20238/com.example.aleal.app_bt_1 D/ViewRootImpl: ViewPostImeInputStage processPointer 0
03-13 08:47:54.913 20238-20238/com.example.aleal.app_bt_1 D/ViewRootImpl: ViewPostImeInputStage processPointer 1
03-13 08:47:54.983 20238-20238/com.example.aleal.app_bt_1 D/BluetoothAdapter: cancelDiscovery
03-13 08:47:54.983 20238-20238/com.example.aleal.app_bt_1 D/BluetoothAdapter: cancelDiscovery = true
03-13 08:47:55.003 20238-20238/com.example.aleal.app_bt_1 D/BLE: PAIR RECEIVER
03-13 08:47:55.913 20238-20238/com.example.aleal.app_bt_1 D/BLE: PAIR RECEIVER
03-13 08:48:05.763 20238-20238/com.example.aleal.app_bt_1 D/ViewRootImpl: ViewPostImeInputStage processPointer 0
03-13 08:48:05.843 20238-20238/com.example.aleal.app_bt_1 D/ViewRootImpl: ViewPostImeInputStage processPointer 1
03-13 08:48:05.913 20238-20308/com.example.aleal.app_bt_1 D/BluetoothAdapter: cancelDiscovery
03-13 08:48:05.923 20238-20308/com.example.aleal.app_bt_1 D/BluetoothAdapter: cancelDiscovery = true
03-13 08:48:05.923 20238-20308/com.example.aleal.app_bt_1 D/BluetoothUtils: isSocketAllowedBySecurityPolicy start : device null
03-13 08:48:05.923 20238-20308/com.example.aleal.app_bt_1 D/BluetoothSocket: connect(): myUserId = 0
03-13 08:48:05.923 20238-20308/com.example.aleal.app_bt_1 W/BluetoothAdapter: getBluetoothService() called with no BluetoothManagerCallback

basel93natour, 2018-03-20:
i have the same problem please  if you come a cross any solution please post  it here

arnleal09, 2018-06-06:
> did anyone find the solution for this issue ? I'm having this error too
> I'm using the library now @basel93natour
> <https://github.com/basel93natour> @arnleal09

This library no support ble devices, to make a connection to ble devices
you need implement you our funtions, go to android documentation to do so.
This library is still usefull to search devices and control bluetooth on
you phone.

proazfar, 2018-07-21:
Is there any version of this library which solves this issue?

OmarAflak, 2018-07-22:
@proazfar @badarshahzad @basel93natour @Amirhgh74 All my apologies for the very late response. As said @arnleal09, this library does not support BLE devices. If time allows it, I will try to implement BLE.
Meanwhile, you can check the Android Documentation : 

https://developer.android.com/guide/topics/connectivity/bluetooth-le

Thank you all for the interest you give to this repository.

closed 2018-07-22 by OmarAflak

vnd-ranga, 2019-06-17:
By adding filter action my problem resolved

     // Register for broadcasts when a device is discovered
        IntentFilter intentFilter = new IntentFilter();
        intentFilter.addAction(BluetoothDevice.ACTION_FOUND);
        intentFilter.addAction(BluetoothAdapter.ACTION_DISCOVERY_STARTED);
        intentFilter.addAction(BluetoothAdapter.ACTION_DISCOVERY_FINISHED);
        registerReceiver(mReceiver, intentFilter);

umamakhalid, 2020-10-22:
If you have tried connecting devices once or more then your socket might be already connected. try restarting devices.