omaraflak/Bluetooth-Library #18: Bluetooth.getDevice is not available in update version

badarshahzad, 2018-03-15:
Hi 
I updated to the new version and I made changes what you refacter the name of methods. But I found in updated version the Bluetooth object doesn't have .getDevice method. Could you tell me how I can get this method call?

OmarAflak, 2018-03-16:
Hi.
Indeed I removed it in recent versions of the library. Unfortunately I can't change the code right now, I'm having troubles with my computer.

What you can do is to keep track of the `BluetoothDevice` object in `onDevicePaired` callback.

badarshahzad, 2018-03-16:
You last version working perfect except the searching but you made the library now bit different now I am facing difficulty to integrate with it. :+1:

closed 2019-05-13 by OmarAflak