omaraflak/Bluetooth-Library #19: Q - Compatibility

LinkOut, 2018-03-19:
Hello, congrats for the nice lib.

Does it works with BLE devices aswell? 

Thanks

OmarAflak, 2018-03-19:
Hi, thanks!
Initially I wrote this library to help communication with devices like HC-05/6. Since it's using Android's bluetooth api, I think it should work.

LinkOut, 2018-03-19:
Nice, thanks for the fast answer, not to open a new issue i wrote you down here a issue im facing: 
`bluetooth = new Bluetooth(MainActivity.this);
        if(!bluetooth.isEnabled())`

The error i got is a NullPointerException:
Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothAdapter.isEnabled()' on a null object reference

LinkOut, 2018-03-19:
Basically, i'm unable to call (!bluetooth.isEnabled()) on the onCreate method of the activity, since calling the bluetooth.onStart(); <- Where you initialize the adapter on the onStart() method...

The order is onCreate then onStart... So i will just call the bluetooth.onStart() after the initialization part. :)

OmarAflak, 2018-03-19:
Yes right ^^ Glad you figured it out yourself !

closed 2018-03-19 by OmarAflak