omaraflak/Bluetooth-Library #24: null pointer on bluetooth isEnable

Amirhgh74, 2018-06-04:
I am getting nullPointer on the first line you wrote if (!bluetooth.isEnable) saying that bluetooth adapter is null , what should I do ?

OmarAflak, 2018-06-04:
`bluetooth.onStart()` method should be called first.

Amirhgh74, 2018-06-04:
that was fixed thanks 
but I think I'm missing something 
is there any way I can see the sample application code for scanning and paring ? 

the problem is when I start scanning , after a while onDiscoverFinished is called and no device is found but the sample works correctly

OmarAflak, 2018-06-04:
Great. Yes you can, the sample code is here : 

https://github.com/OmarAflak/Bluetooth-Library/tree/master/app

More specifically here : 

https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ui/scan/interactor/ScanInteractorImpl.java

You have to handle `onDeviceFound()`, it is called every time a device is discovered.

Amirhgh74, 2018-06-04:
I have done what is needed I think 
here is my onCreate function where I'm testing things 

`if (!bluetooth.isEnabled()){
            bluetooth.enable();
        }
        
        bluetooth.setBluetoothCallback(new BluetoothCallback() {
            @Override
            public void onBluetoothTurningOn() {}

            @Override
            public void onBluetoothOn() {
                Log.e("amir" , "bluetooth is turned on !");
            }

            @Override
            public void onBluetoothTurningOff() {}

            @Override
            public void onBluetoothOff() {}

            @Override
            public void onUserDeniedActivation() {
                // when using bluetooth.showEnableDialog()
                // you will also have to call bluetooth.onActivityResult()
            }
        });

        bluetooth.setDiscoveryCallback(new DiscoveryCallback() {
            @Override public void onDiscoveryStarted() {
                Log.e("amir" , "start discovery");
            }
            @Override public void onDiscoveryFinished() {
                Log.e("amir" , "finished discovery");
            }
            @Override public void onDeviceFound(BluetoothDevice device) {
                Log.e("amir" , "we have new device : " + device.getName());
            }
            @Override public void onDevicePaired(BluetoothDevice device) {}
            @Override public void onDeviceUnpaired(BluetoothDevice device) {}
            @Override public void onError(String message) {

                Log.e("amir" , message);

            }
        });

        new Handler().postDelayed(new Runnable() {
            @Override
            public void run() {
                bluetooth.startScanning();
            }
        }, 1000);`

when I start scanning, start discovery and finish discovery is logged correctly but no log for finding devices

OmarAflak, 2018-06-14:
This is a working example, you just need to request the permissions if not granted : 

```
public class MainActivity extends AppCompatActivity implements BluetoothCallback, DiscoveryCallback {
    private Bluetooth bluetooth;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);

        bluetooth = new Bluetooth(this);
        bluetooth.setBluetoothCallback(this);
        bluetooth.setDiscoveryCallback(this);
    }

    @Override
    protected void onStart() {
        super.onStart();
        bluetooth.onStart();
        if(hasPermissions()){
            bluetooth.enable();
            bluetooth.startScanning();
        }
        else{
            // request permissions...
        }
    }

    boolean hasPermissions(){
        return
            ContextCompat.checkSelfPermission(this, Manifest.permission.BLUETOOTH) == PackageManager.PERMISSION_GRANTED &&
            ContextCompat.checkSelfPermission(this, Manifest.permission.BLUETOOTH_ADMIN) == PackageManager.PERMISSION_GRANTED &&
            ContextCompat.checkSelfPermission(this, Manifest.permission.ACCESS_COARSE_LOCATION) == PackageManager.PERMISSION_GRANTED;
    }

    @Override public void onBluetoothTurningOn() {}

    @Override
    public void onBluetoothOn() {
        Log.d(getClass().getSimpleName(), "Bluetooth on");
    }

    @Override public void onBluetoothTurningOff() {}
    @Override public void onBluetoothOff() {}
    @Override public void onUserDeniedActivation() {}

    @Override
    public void onDiscoveryStarted() {
        Log.d(getClass().getSimpleName(), "scanning...");
    }

    @Override
    public void onDiscoveryFinished() {
        Log.d(getClass().getSimpleName(), "scanning finished.");
    }

    @Override
    public void onDeviceFound(BluetoothDevice device) {
        Log.d(getClass().getSimpleName(), "["+device.getAddress()+"] : "+device.getName());
    }

    @Override public void onDevicePaired(BluetoothDevice device) {}
    @Override public void onDeviceUnpaired(BluetoothDevice device) {}
    @Override public void onError(String message) {}
}

```

Amirhgh74, 2018-06-22:
thanks for your help 
i was finally able to get it to work , but i have one more issue , when i use connect to device function after pairing with a device, nothing will get called just this line will be logged in logcat 
`getBluetoothService() called with no BluetoothManagerCallback`
i searched and found that this is not an error its just a warning but i dont get any error on connection and my log in `onDeviceConnected` function wont get called either 
how can i get the status to check if im connected to the device ?

hossein-hbb, 2018-07-27:
Hello omar .
how can i use DeviceCallback and receive message from Bluetooth?

hossein-hbb, 2018-08-07:
Hi ho to use (Listen on bluetooth socket) inside fragment?

hossein-hbb, 2018-08-07:
Caused by: java.lang.ClassCastException: fragments.RemotePanelFragment cannot be cast to fragments.SearchBTFragment

closed 2019-05-13 by OmarAflak