omaraflak/Bluetooth-Library #26: Possible memory leak on Dexter

mr0bles, 2018-06-12:
We're having a memory leak using Dexter, do you have any idea of fixing it? 

Thanks!

[image]

OmarAflak, 2018-06-12:
Sorry I don't. I actually stopped using Dexter, it turns out it's pretty simple to implement permission requests on Android.
https://developer.android.com/training/permissions/requesting#java

closed 2018-06-12 by OmarAflak

mr0bles, 2018-06-12:
@OmarAflak 
I checked that you solved some leaks (receivers) in the current code, when you going to publish new version?

OmarAflak, 2018-06-12:
I updated the library, it should be available now.

`implementation 'me.aflak.libraries:bluetooth:1.3.4'`

mr0bles, 2018-06-12:
I have a leak on `bluetooth.startScanning();` when close (or press back) when i scanning

i'll try solved it to help you

`06-12 17:56:24.887 2424-2424/xx.xxxxxxx.xxx E/ActivityThread: Activity xx.xxxxxxx.xxx.ui.lector.scan.view.ScanActivity has leaked IntentReceiver me.aflak.bluetooth.Bluetooth$6@117f19f8 that was originally registered here. Are you missing a call to unregisterReceiver()?
    android.app.IntentReceiverLeaked: Activity xx.xxxxxxx.xxx.ui.lector.scan.view.ScanActivity has leaked IntentReceiver me.aflak.bluetooth.Bluetooth$6@117f19f8 that was originally registered here. Are you missing a call to unregisterReceiver()?
        at android.app.LoadedApk$ReceiverDispatcher.<init>(LoadedApk.java:1015)
        at android.app.LoadedApk.getReceiverDispatcher(LoadedApk.java:802)
        at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:2078)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:2058)
        at android.app.ContextImpl.registerReceiver(ContextImpl.java:2052)
        at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:504)
        at me.aflak.bluetooth.Bluetooth.startScanning(Bluetooth.java:217)
        at xx.xxxxxxx.xxx.ui.lector.scan.interactor.ScanInteractorImpl.scanDevices(ScanInteractorImpl.java:94)
        at xx.xxxxxxx.xxx.ui.lector.scan.presenter.ScanPresenterImpl.startScanning(ScanPresenterImpl.java:46)
        at xx.xxxxxxx.xxx.ui.lector.scan.presenter.ScanPresenterImpl.onStart(ScanPresenterImpl.java:27)
        at xx.xxxxxxx.xxx.ui.lector.scan.view.ScanActivity.onStart(ScanActivity.java:146)
        at android.app.Instrumentation.callActivityOnStart(Instrumentation.java:1264)
        at android.app.Activity.performStart(Activity.java:6646)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3096)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3243)
        at android.app.ActivityThread.access$1000(ActivityThread.java:218)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1718)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:145)
        at android.app.ActivityThread.main(ActivityThread.java:6914)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:1404)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1199)`

OmarAflak, 2018-06-12:
You have to call `bluetoothOnStop();`  before quitting