omaraflak/Bluetooth-Library #30: Kotlin: No device is being discovered, onDeviceFound never invoked

thepipeman, 2018-10-11:
I am using the library and I'm implementing it using Kotlin, everything is fine when I do scan, stop scan, I have an instance of the `Bluetooth` class however, I never get to discover new devices and pair to any device

I created a very simple activity, the code is below, I only Toast when a callback is invoked

```
class MainActivity : AppCompatActivity(), DiscoveryCallback, BluetoothCallback, DeviceCallback {

    private lateinit var bluetooth: Bluetooth
    private var devices: MutableList<BluetoothDevice> = ArrayList()

    override fun onCreate(savedInstanceState: Bundle?) {
        super.onCreate(savedInstanceState)
        setContentView(R.layout.activity_main)
        setSupportActionBar(toolbar)

        bluetooth = Bluetooth(this)
        bluetooth.onStart()
        bluetooth.setCallbackOnUI(this)
        bluetooth.setDiscoveryCallback(this)
        bluetooth.setBluetoothCallback(this)

        devices.addAll(BluetoothAdapter.getDefaultAdapter().bondedDevices)

        fab.setOnClickListener { view ->
            // to scan uncomment this code
             bluetooth.startScanning()
            // when scanning onDeviceFound callback should be invoked, but for some reasons, it doesn't work when it is being used in Kotlin

            // to pair to the connected device, I was only paired to once device so please change it  accordingly.
//            bluetooth.pair(devices[0])
        }
    }

    override fun onCreateOptionsMenu(menu: Menu): Boolean {
        menuInflater.inflate(R.menu.menu_main, menu)
        return true
    }

    override fun onOptionsItemSelected(item: MenuItem): Boolean {
        return when (item.itemId) {
            R.id.action_settings -> true
            else -> super.onOptionsItemSelected(item)
        }
    }

    override fun onDevicePaired(device: BluetoothDevice?) {
        Toast.makeText(this, "onDevicePaired", Toast.LENGTH_SHORT).show()
    }

    override fun onDiscoveryStarted() {
        Toast.makeText(this, "onDiscoveryStarted", Toast.LENGTH_SHORT).show()
    }

    override fun onDeviceUnpaired(device: BluetoothDevice?) {
        Toast.makeText(this, "onDeviceUnpaired", Toast.LENGTH_SHORT).show()
    }

    override fun onError(message: String?) {
        Toast.makeText(this, "onError", Toast.LENGTH_SHORT).show()
        Log.d("OnError", message)
    }

    override fun onDiscoveryFinished() {
        Toast.makeText(this, "onDiscoveryFinished", Toast.LENGTH_SHORT).show()
    }

    override fun onDeviceFound(device: BluetoothDevice?) {
        Toast.makeText(this, "onDeviceFound ${device!!.name}", Toast.LENGTH_SHORT).show()
    }

    override fun onUserDeniedActivation() {
        Toast.makeText(this, "onUserDeniedActivation", Toast.LENGTH_SHORT).show()
    }

    override fun onBluetoothOff() {
        Toast.makeText(this, "onBluetoothOff", Toast.LENGTH_SHORT).show()
    }

    override fun onBluetoothOn() {
        Toast.makeText(this, "onBluetoothOn", Toast.LENGTH_SHORT).show()
    }

    override fun onBluetoothTurningOn() {
        Toast.makeText(this, "onBluetoothTurningOn", Toast.LENGTH_SHORT).show()
    }

    override fun onBluetoothTurningOff() {
        Toast.makeText(this, "onBluetoothTurningOff", Toast.LENGTH_SHORT).show()
    }

    override fun onDeviceDisconnected(device: BluetoothDevice?, message: String?) {
        Toast.makeText(this, "onDeviceDisconnected", Toast.LENGTH_SHORT).show()
    }

    override fun onDeviceConnected(device: BluetoothDevice?) {
        Toast.makeText(this, "onDeviceConnected ${device!!.name}", Toast.LENGTH_SHORT).show()
    }

    override fun onConnectError(device: BluetoothDevice?, message: String?) {
        Toast.makeText(this, "onConnectError", Toast.LENGTH_SHORT).show()
    }

    override fun onMessage(message: String?) {
        Toast.makeText(this, "onMessage", Toast.LENGTH_SHORT).show()
    }
}```

`bluetooth.startScanning` or `bluetooth.pair()` should invoke the callback properly, however for some reasons it fails on my Kotlin implementation.

Is anything wrong in this one?

Thank you!

akimaleo, 2018-10-17:
> The other permission that you must declare is either ACCESS_COARSE_LOCATION or ACCESS_FINE_LOCATION. A location permission is required because Bluetooth scans can be used to gather information about the location of the user. This information may come from the user's own devices, as well as Bluetooth beacons in use at locations such as shops and transit facilities.

From Android Marshmallow or above you should declare and permit <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/> for application.

Read article below.
https://developer.android.com/guide/topics/connectivity/bluetooth

OmarAflak, 2019-05-13:
@akimaleo is right. And btw, you don't need to call `BluetoothAdapter.getDefaultAdapter().bondedDevices`, just use `bluetooth.getPairedDevices()`

closed 2019-05-13 by OmarAflak