omaraflak/Bluetooth-Library #32: Can we create an insecure connection with this library ?

YushaBinArif, 2019-03-10:
I used the following line of code with native Bluetooth library to create an insecure connection with HC-05(a Bluetooth module for Arduino) can I do same with this library?

BA = BluetoothAdapter.getDefaultAdapter();//get the mobile bluetooth device
//connects to the device's MAC_ADDRESS and checks if it's available
BluetoothDevice BD = BA.getRemoteDevice(MAC_ADDRESS);                    
BS = BD.createInsecureRfcommSocketToServiceRecord(myUUID);
//create a RFCOMM (SPP) connection
BluetoothAdapter.getDefaultAdapter().cancelDiscovery();
BS.connect();

OmarAflak, 2019-03-11:
Hi @YushaBinArif, yes this is how the library works.

YushaBinArif, 2019-03-13:
Hi, Thanks @OmarAflak  for replying to my query.

closed 2019-03-13 by YushaBinArif