omaraflak/Bluetooth-Library #36: Sending raw buffer (0 - 255)

dobraMorda, 2019-05-10:
Hello!

I need to send over bluetooth spp raw buffer (variables with value range: 0 - 255). Is this possible? I was looking for some method like "bluetooth.write(buffer, len)", but nothing to found...

Best regards!
Patryk

dobraMorda, 2019-05-12:
Solved.

`byte[] bytes = new byte[ n ];
byte val = 0;
		for ( int i = 0 ; i < n ; i++ ) {
			val = ( byte ) ( buffer[ i ] & 0xFF );
			bytes[ i ] = val;
		}
bluetooth.getSocket ( ).getOutputStream ( ).flush ( );
bluetooth.getSocket ( ).getOutputStream ( ).write ( bytes, 0, 256 );
`

Masking bytes and send it over socket output stream.

closed 2019-05-12 by dobraMorda

OmarAflak, 2019-05-12:
Hey @dobraMorda, thanks for posting your answer here ! I'm glad you found a solution !