omaraflak/Bluetooth-Library #38: Attempt to invoke virtual method 'java.io.OutputStream me.aflak.bluetooth.Bluetooth$ReceiveThread.getOutputStream()' on a null object reference

kvnramirez, 2019-05-30:
Hi, I'm trying to send a message inmediately after the device is connected, but keeps throwing this error: 

java.lang.NullPointerException: Attempt to invoke virtual method 'java.io.OutputStream me.aflak.bluetooth.Bluetooth$ReceiveThread.getOutputStream()' on a null object reference

I'm sending it on DeviceCallback  using onDeviceConnected function

override fun onDeviceConnected(device: BluetoothDevice) {
            statusMessage?.text = getString(R.string.connected)
                    bluetooth?.send("3")
        }

I need to sent the value inmediately after connection to start to read sensors from an arduino

OmarAflak, 2019-05-31:
Hi, please post the code of your Activity. Also take a look at that example : [https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ChatActivity.java](https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ChatActivity.java)

OmarAflak, 2019-07-07:
This issue was solved in the new version :

```
implementation 'me.aflak.libraries:bluetooth:1.3.7'
```

closed 2019-07-07 by OmarAflak