omaraflak/Bluetooth-Library #39: connection error

sontqq, 2019-06-05:
Everytime I try to connect to a device i get:

`error_read failed, socket might closed or timeout, read ret: -1`

Any idea?

OmarAflak, 2019-06-05:
What kind of the device are you trying to connect to ?

sontqq, 2019-06-05:
I'm trying to connect from an Android 9.0 Samsung device to a Windows 10 PC with a bluetooth dongle.

I have also tried to connect with the port tricking but no success.

sontqq, 2019-06-06:
I get this error:
```
W/System.err: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothSocket android.bluetooth.BluetoothDevice.createRfcommSocketToServiceRecord(java.util.UUID)' on a null object reference
        at me.aflak.bluetooth.Bluetooth.connect(Bluetooth.java:462)
W/System.err:     at me.aflak.bluetooth.Bluetooth.connectToDevice(Bluetooth.java:271)
        at me.aflak.bluetooth.Bluetooth.connectToDevice(Bluetooth.java:279)
W/System.err:     at com.sontme.esp.getlocation.activities.MainActivity$3$2.onDiscoveryFinished(MainActivity.java:363)
        at me.aflak.bluetooth.Bluetooth$7$3.run(Bluetooth.java:597)
        at me.aflak.bluetooth.utils.ThreadHelper.run(ThreadHelper.java:15)
W/System.err:     at me.aflak.bluetooth.Bluetooth$7.onReceive(Bluetooth.java:594)
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$getRunnable$0(LoadedApk.java:1486)
        at android.app.-$$Lambda$LoadedApk$ReceiverDispatcher$Args$_BumDX2UKsnxLVrE6UJsJZkotuA.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:873)
W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7073)
W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)
```

jpeiter, 2020-05-19:
I'm having the same error, using either `connectToDevice` and `connectToDeviceWithPortTrick`, but fails every time.
I downloaded the [Bluetooth Terminal app](https://play.google.com/store/apps/details?id=me.aflak.bluetoothterminal) and it logs the same error on the screen.

glebkravchenko, 2020-08-20:
yep, same for me, trying to connect to medical device, but it is saying same, error_read failed, socket might closed or timeout, read ret: -1

OmarAflak, 2020-08-20:
Please make sure you are using a regular Bluetooth dongle, not BLE. BLE is not supported.

charlieporth1, 2021-12-20:
I'm getting this msg too. Is there any solution? Is there anyway to disable BLE @OmarAflak?

ibrahim-iqbal, 2022-01-10:
This is issue is still persistent, is there a workaround for this, none of the other libraries are also able to tackle this.

skillhubz, 2023-10-16:
> Everytime I try to connect to a device i get:
> 
> `error_read failed, socket might closed or timeout, read ret: -1`
> 
> Any idea?

no

skillhubz, 2023-10-16:
> Please make sure you are using a regular Bluetooth dongle, not BLE. BLE is not supported.

gotcha