omaraflak/Bluetooth-Library #40: Attempt to invoke virtual method 'android.bluetooth.BluetoothSocket me.aflak.bluetooth.Bluetooth$ReceiveThread.getSocket()' on a null object reference

ScoutHUD, 2019-07-06:
Hi @OmarAflak , I am seeing an error when trying to use `bluetooth.getSocket()` after successfully connecting, I keep getting the following error: 

```
W/System.err: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothSocket me.aflak.bluetooth.Bluetooth$ReceiveThread.getSocket()' on a null object reference
        at me.aflak.bluetooth.Bluetooth.getSocket(Bluetooth.java:144)
        at com.scout.eurobeatandroid.MainActivity$9.onDeviceConnected(MainActivity.java:207)
        at me.aflak.bluetooth.Bluetooth$6$1.run(Bluetooth.java:485)
        at me.aflak.bluetooth.utils.ThreadHelper.run(ThreadHelper.java:15)
        at me.aflak.bluetooth.Bluetooth$6.run(Bluetooth.java:482)
07-06 21:15:10.697 7745-7795/com.scout.ccc W/System.err:     at java.lang.Thread.run(Thread.java:764)
```

I have have copied the code examples correctly, so i'm not sure if there is an issue with the method, or I am doing something wrong.

```
// CONNECTION
private void connect(String name) {
     Toast.makeText(getApplicationContext(),"Attempting to Connect to " + name, Toast.LENGTH_SHORT).show();
     bluetooth.connectToName(name);
}

// ON DEVICE CONNECTION CALLBACK
      bluetooth.setDeviceCallback(new DeviceCallback() {
            @Override public void onDeviceConnected(BluetoothDevice device) {
                runOnUiThread(new Runnable() {public void run() {final Toast toast = Toast.makeText(getApplicationContext(), "Connected to OBD-II", Toast.LENGTH_SHORT);
                    toast.show();
                    bluetoothPairedList.setVisibility(View.GONE);
                }});

                // setup OBD-11 params
                BluetoothSocket socket = bluetooth.getSocket(); // NULL POINTER EXCEPTION HERE

                try {
                    new EchoOffCommand().run(socket.getInputStream(), socket.getOutputStream());
                    new LineFeedOffCommand().run(socket.getInputStream(), socket.getOutputStream());
                    new TimeoutCommand(125).run(socket.getInputStream(), socket.getOutputStream());
                    new SelectProtocolCommand(ObdProtocols.AUTO).run(socket.getInputStream(), socket.getOutputStream());
                } catch (final Exception e) {
                    // handle errors
                    e.printStackTrace();
                }
            }

...

}
```

Thanks for taking a look.

OmarAflak, 2019-07-07:
There's indeed a problem with the library, thanks for letting me know.

I'll update the library today. I'll keep you updated. In the meantime, a dirty fix would be to wait in a thread 3sec and then do your job (still in the thread).

OmarAflak, 2019-07-07:
I've updated the library, this should no longer be a problem now.

```
implementation 'me.aflak.libraries:bluetooth:1.3.7'
```

closed 2019-07-07 by OmarAflak

harnie2275, 2022-01-12:
Please where are we to but this implementation 'me.aflak.libraries:bluetooth:1.3.7'