omaraflak/Bluetooth-Library #41: String Translate ...

Deltafox79, 2019-07-18:
Hi, Is Possible to Set the Custom String text for the Word : "Scanning...", "Pairing...", "Done" for the layout Id **activity_scan_state** ?

Many Thank's

OmarAflak, 2019-07-18:
Seems like the code you are looking at is simply the example code that come along with the library. This is just meant to show you how to use the library, you can use it however you want. To use the library in your Android project, simply add the following line in your build.gradle dependencies section :

```
implementation 'me.aflak.libraries:bluetooth:1.3.7'
```

Hit me up if you have any problem. Also take a look at the README, should help you.

Deltafox79, 2019-07-18:
My Fault.. :-) Thank's for everything..

closed 2019-07-18 by Deltafox79