omaraflak/Bluetooth-Library #44: Allow sending and receiving data (bytes) without with use of CR LF delimiters

alanoatwork, 2019-08-07:


alanoatwork, 2019-08-07:
I need to send binary data as a packet, so the CR LF restriction is limiting. Please allow the option to remove this restriction.

OmarAflak, 2019-08-11:
Hey @alanoatwork, thanks for your suggestion. The send part of okay, but how would you then receive data on Android side ? How should we read from the InputStream ?

OmarAflak, 2019-08-11:
Okay I've fixed it. Sending is not a problem, you can now send messages as `String` or `byte[]`. In order to receive message you have 2 options :
1) Use the default behavior that is going to read until it hits a new line, then return the message in a byte array;
2) Create your own reader class, it should extend from `SocketReader`;

The first option is actually a particular case of the second one  : [https://github.com/OmarAflak/Bluetooth-Library/blob/issue_44/bluetooth/src/main/java/me/aflak/bluetooth/reader/LineReader.java](https://github.com/OmarAflak/Bluetooth-Library/blob/issue_44/bluetooth/src/main/java/me/aflak/bluetooth/reader/LineReader.java)

OmarAflak, 2019-08-11:
Unfortunately I can't test the code as I don't have any hardware right now, therefore I can't release the new version. If you want to test it yourself however, here is the new branch : [https://github.com/OmarAflak/Bluetooth-Library/tree/issue_44](https://github.com/OmarAflak/Bluetooth-Library/tree/issue_44)

alanoatwork, 2019-08-11:
Omar,

 

Thanks for updating your library!

 

After adding:

maven { url "https://jitpack.io" }

 

to my project build.gradle, and adding:

implementation 'com.github.OmarAflak:Bluetooth-Library:issue_44-SNAPSHOT'

 

to my app build.gradle I was able to rebuild and try the library.

 

I have an HC-06 bluetooth module connected with the TX and RX lines tied together, so that module should simply echo back any characters it receives.  I sent the following data in debugger mode from my main activity:

 

@Override

    public boolean onOptionsItemSelected(MenuItem item) {

        switch(item.getItemId()) {

            case R.id.send_data:

                msg("Pressed send data.");

                byte data[] = {0,1,2,3,4,5};

                mBluetooth.send(data);

                return true;

        }

        return super.onOptionsItemSelected(item);

    }

 

However, there’s no callback on:

 

public void onMessage(byte[] message) {
   String s = String.format("%d", message[0]);
    Log.d(TAG, s);
}

 

I’m guessing that’s because you said I needed to , “Create your own reader class, it should extend from SocketReader”.

 

So I could use some help here as I’m a novice Android/Java programmer. Should I create a new class based on your library?  I tried creating a new class called MyBluetooth:

 

package com.ninecmed.tablet;

 

import me.aflak.bluetooth.Bluetooth;

import me.aflak.bluetooth.reader.SocketReader;

 

public class MyBluetooth extends Bluetooth,SocketReader {

 

};

 

But AndroidStudio gives me an error saying that there’s no default constructor available in me.aflak.bluetooth.Bluetooth

 

Could you explain how I should extend SocketReader so that I’ll receive a callback on OnMessage?  

 

Regards,

Alan

OmarAflak, 2019-08-12:
Hey Alan, this issue is fixed in the new version.

```
implementation 'me.aflak.libraries:bluetooth:1.3.8'
```

To have more control over the socket reader, you should create a class that extends `SocketReader` and override `byte[] read() throw IOException`. This is an example :

```java
public class CustomReader extends SocketReader {
    private Scanner scanner;

    public DelimiterReader(InputStream inputStream) {
        super(inputStream);
        scanner = new Scanner(inputStream);
        scanner.useDelimiter("\0"); // set your delimiter here
    }

    @Override
    public byte[] read() throws IOException {
        IOException e = scanner.ioException();
        if(e != null){
            throw e;
        }
        return scanner.next().getBytes();
    }
}
```

Then,

```java
bluetooth.setReader(CustomReader.class)
```

closed 2019-08-12 by OmarAflak