omaraflak/Bluetooth-Library #46: NullPointerException when starting another activity

alanoatwork, 2019-09-05:
Omar,  if I'm not connected to my bluetooth device and I launch another activity, I get the following error:

```
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.ninecmed.tablet, PID: 5011
    java.lang.RuntimeException: Unable to stop activity {com.ninecmed.tablet/com.ninecmed.tablet.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothSocket me.aflak.bluetooth.Bluetooth$ReceiveThread.getSocket()' on a null object reference
        at android.app.ActivityThread.performStopActivityInner(ActivityThread.java:3360)
        at android.app.ActivityThread.handleStopActivity(ActivityThread.java:3406)
        at android.app.ActivityThread.access$1100(ActivityThread.java:147)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1310)
        at android.os.Handler.dispatchMessage(Handler.java:102)
        at android.os.Looper.loop(Looper.java:135)
        at android.app.ActivityThread.main(ActivityThread.java:5253)
        at java.lang.reflect.Method.invoke(Native Method)
        at java.lang.reflect.Method.invoke(Method.java:372)
        at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:899)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:694)
     Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.bluetooth.BluetoothSocket me.aflak.bluetooth.Bluetooth$ReceiveThread.getSocket()' on a null object reference
        at me.aflak.bluetooth.Bluetooth.disconnect(Bluetooth.java:297)
        at com.ninecmed.tablet.MainActivity.onStop(MainActivity.java:142)
        at android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1261)
        at android.app.Activity.performStop(Activity.java:6085)
        at android.app.ActivityThread.performStopActivityInner(ActivityThread.java:3357)
```
Am I doing something incorrect?  
Cheers!

OmarAflak, 2019-09-05:
Hey Alan,

Are you calling getSocket() or something in the onStop() method of  your MainActivity.java ?

alanoatwork, 2019-09-05:
Omar,
I shouldn't code late a night!  I was calling disconnect() without checking isConnected() in the method onStop()!  All is well now.  I've been exercising the library for weeks now and other than my mistake, I haven't seen any other issues!
Cheers,
Alan

closed 2019-09-05 by alanoatwork

OmarAflak, 2019-09-05:
Great ! I'm glad you fixed it :)
Ah good to know ! Hit me up anytime !