omaraflak/Bluetooth-Library #49: About scanning the nearby bluetooth device

JMJ15, 2019-12-21:
Hey, I try to use your library to connect Android and Arduino. I am able to connect already paired  device and also able to send and receive data between Android and Arduino. But facing problem to scan nearby device and list it. Even the progress bar is simply rotating not working the function setProgressAndState().  My keen guess is 
 private DiscoveryCallback discoveryCallback = new DiscoveryCallback() {......}
is not working or when the scan button is pressed, the DiscoveryCallback is not calling.
I followed the code as it is in the link mentioned below
 https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ScanActivity.java

OmarAflak, 2019-12-21:
Hey, when you start the app do you have any logs ? Also is your settings app detecting your device ? Can you try to see if this app works properly on your device https://play.google.com/store/apps/details?id=me.aflak.bluetoothterminal

JMJ15, 2019-12-23:
I don't have any logs. And my settings app detecting the other discovered devices and listing there.  I try with this app https://play.google.com/store/apps/details?id=me.aflak.bluetoothterminal but no result.

OmarAflak, 2019-12-23:
What bluetooth module are you trying to connect to ?

JMJ15, 2019-12-23:
HC05 bluetooth module. I found problem only at scanning part. The app is listing already paired devices and able to connect with those device and able to send and receive data. But when button is pressed for scanning, it is not working. I used try catch method to verify whether bluetooth.startScanning(); is working. It was working but  private DiscoveryCallback discoveryCallback = new DiscoveryCallback() {......} is not working. I don't know where it is getting problem.

JMJ15, 2020-01-06:
When I try with https://play.google.com/store/apps/details?id=me.aflak.libraries&hl=en_IN and https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_IN , both apps are able to do scanning nearby available bluetooth device and connect with those. So what is missing with this https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ScanActivity.java programming side?

Even when I open https://play.google.com/store/apps/details?id=me.aflak.libraries&hl=en_IN and do the scanning for new device and immediatiley opening my app (which I programmed using  https://github.com/OmarAflak/Bluetooth-Library/blob/master/app/src/main/java/me/aflak/libraries/ScanActivity.java ) then only the progress bar message of below mentioned sections are working, but not the scannedDevices = new ArrayList<>(); listing in my app.
 
 @Override
        public void onDiscoveryStarted() {
            setProgressAndState("Scanning...", View.VISIBLE);
            scannedDevices = new ArrayList<>();
            scanning = true;
        }

        @Override
        public void onDiscoveryFinished() {
            setProgressAndState("Done.", View.VISIBLE);
            scanning = false;
        }

OmarAflak, 2020-01-06:
Okay. I'll have a look later today.

OmarAflak, 2020-01-09:
Hi @JMJ15. It seems that this is caused by the location of the device not being enabled.

closed 2020-01-09 by OmarAflak