omaraflak/Bluetooth-Library #51: onMessage method is not calling after device sending data to application

sagarchorage, 2020-02-24:
Hello team,

I have implemented this library for sharing data.
but after sending data to device :
---------------
String s = "*" +"Log_Interval," + selectedValue + "," +  "&";
                        bluetooth.send(s);
--------------
its properly sending data to that device but after device responds with success but not getting to our application (onMessage or onError method is not calling)

please help me to identify this issue what actually I did mistake in my code while using library

OmarAflak, 2020-02-24:
Hi @sagarchorage, this is probably because of how you are sending the message from the Bluetooth device (not Android). You need to add a `\n` at the end of your message so that the Android knows when to stop reading from the input stream. Keep me updated!

sagarmobicode, 2020-02-24:
Thanks, @OmarAflak for your valuable response
After adding "\n" in the end string issue got resolved.

Thanks once again 

Thanks & Regards,
Sagar C.

closed 2020-02-24 by OmarAflak