omaraflak/Bluetooth-Library #55: Can not terminate socket on activity close.

amit-jangir123, 2020-08-07:
@Override
    public void onConnect(BluetoothDevice device) {
        Display("Connected to "+device.getName()+" - "+device.getAddress());
        this.runOnUiThread(new Runnable() {
            @Override
            public void run() {

            }
        });
    }

    public void Display(final String s){
        this.runOnUiThread(new Runnable() {
            @Override
            public void run() {
                Log.e("Status","------------------------------------------"+s);
            }
        });
    }

    @Override
    public void onDisconnect(BluetoothDevice device, String message) {
        Display("Disconnected!");
        Display("Connecting again...");
        b.connectToDevice(device);
    }

    @Override
    public void onMessage(String message) {

        this.runOnUiThread(new Runnable() {
            @Override
            public void run() {
                String str = message;
                str = str.substring(1,str.length());
                String liter = str.substring(0,3);
                String subleter = str.substring(4,5);
                String lastLeter = liter+"."+subleter;

                Log.e("Message","------------------------------------------"+message);
                if(et_weightHasFocus && message.contains("L"))
                    et_weight.setText(lastLeter);
            }
        });

    }

    @Override
    public void onError(String message) {
        Display("Error: "+message);
    }

    @Override
    public void onConnectError(final BluetoothDevice device, String message) {
        Display("Error: "+message);
        Display("Trying again in 3 sec.");
        runOnUiThread(new Runnable() {
            @Override
            public void run() {
                Handler handler = new Handler();
                handler.postDelayed(new Runnable() {
                    @Override
                    public void run() {
                        b.connectToDevice(device);
                    }
                }, 2000);
            }
        });
    }

private final BroadcastReceiver mReceiver = new BroadcastReceiver() {
        @Override
        public void onReceive(Context context, Intent intent) {
            final String action = intent.getAction();

            if (action.equals(BluetoothAdapter.ACTION_STATE_CHANGED)) {
                final int state = intent.getIntExtra(BluetoothAdapter.EXTRA_STATE, BluetoothAdapter.ERROR);
             //   Intent intent1 = new Intent(Chat.this, Select.class);

                switch (state) {
                    case BluetoothAdapter.STATE_OFF:
                        if(registered) {
                            unregisterReceiver(mReceiver);
                            registered=false;
                        }
//                        startActivity(intent1);
//                        finish();
                        break;
                    case BluetoothAdapter.STATE_TURNING_OFF:
                        if(registered) {
                            unregisterReceiver(mReceiver);
                            registered=false;
                        }
                  //      startActivity(intent1);
                        finish();
                        break;
                }
            }
        }
    };

 @Override
    public void onBackPressed() {

        Log.e("register",registered+"");
        if(registered) {
            unregisterReceiver(mReceiver);
            registered=false;
        }

        finish();
    }

I have to close the socket because next time i open it, it fails to connect Says Try again in 3 seconds
Need Help

amit-jangir123, 2020-08-07:
Same happing in your code too, socket is not terminating

OmarAflak, 2020-08-08:
You don't have to register and unregister broadcast listeners yourself. Call `disconnect()` and `onStop()`