omaraflak/Bluetooth-Library #56: startDiscovery impossible

openmotion, 2020-08-20:
Hello

i try to use your library with a simple code
 bluetooth = new Bluetooth(this);

        bluetooth.setBluetoothCallback(bluetoothCallback);
        bluetooth.setDiscoveryCallback(discoveryCallback);

        if(bluetooth != null) {
            try {
                bluetooth.startScanning();
            } catch (Exception e) {
                e.printStackTrace();
            }
        }

but i have this error

 java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.bluetooth.BluetoothAdapter.startDiscovery()' on a null object reference
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at me.aflak.bluetooth.Bluetooth.startScanning(Bluetooth.java:381)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at fr.openmotion.lvd.gui.bluetoothaflak.discovery.onCreate(discovery.java:65)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.Activity.performCreate(Activity.java:7087)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.Activity.performCreate(Activity.java:7078)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1215)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2763)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2888)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.ActivityThread.-wrap11(Unknown Source:0)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1609)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:106)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.os.Looper.loop(Looper.java:171)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at android.app.ActivityThread.main(ActivityThread.java:6606)
2020-08-20 17:00:21.092 26670-26670/fr.openmotion.lvd W/System.err:     at java.lang.reflect.Method.invoke(Native Method)
2020-08-20 17:00:21.093 26670-26670/fr.openmotion.lvd W/System.err:     at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:518)
2020-08-20 17:00:21.093 26670-26670/fr.openmotion.lvd W/System.err:     at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:823)

openmotion, 2020-08-20:
hello it's ok but now i have this error when i try to connect : Could not connect, next try in 3 sec... : read failed, socket might closed or timeout, read ret: -1
i have the same error in your demo app :p

OmarAflak, 2020-08-20:
Is the device you are trying to connect to a BLE device ?

openmotion, 2020-08-20:
> Is the device you are trying to connect to a BLE device ?

Yes it's possible
I can connect to the device with the default tchat app from Google. I am searching for a library able to manage jpg file transfer

openmotion, 2020-08-21:
> > Is the device you are trying to connect to a BLE device ?
> 
> Yes it's possible
> I can connect to the device with the default tchat app from Google. I am searching for a library able to manage jpg file transfer

The device must be BLE compatible yes but i don't want to connect it in BLE

OmarAflak, 2020-08-21:
The library does not support BLE.
Of you run the app in the repo does it connect without problem ?
Socket timeout might happen few times before connecting successfully..

openmotion, 2020-08-21:
> The library does not support BLE.
> Of you run the app in the repo does it connect without problem ?
> Socket timeout might happen few times before connecting successfully..

the app in the repo does not connect :p