omaraflak/Bluetooth-Library #58: Connected but chat not working

Ted-kim-dev, 2020-08-30:
I'm not able to receive data on the other end
Data cannot be sent to the device too.
module hc-06

Ted-kim-dev, 2020-08-30:
module hc-06
bluetooth version 2.x

OmarAflak, 2020-08-30:
Regarding the device side: the default behaviour is to read data line by line (you can change that, see README). So if you want to receive data on your phone you should send data with "\n" at the end.

Regarding HC-06 side: I don't know why cannot receive data if you managed to connect. Can you send the receiving code ? Are you using Arduino ?

Ted-kim-dev, 2020-08-31:
Regarding the device side: I tried the solution, but it didn't work.
Regarding HC-06 side: 
I'm using Arduino.
Do I have to set the tx / rx value?