omaraflak/Bluetooth-Library #59: SocketReader read() returning null

nameispriyank, 2020-09-08:
Hi @OmarAflak I had experienced a crash in the app pointing to the library's Socket Reader `read()` function. I was wondering if you can clarify why do we return null by default for the `read()` function [here](bluetooth/src/main/java/me/aflak/bluetooth/reader/SocketReader.java)

`public byte[] read() throws IOException {
        return null;
 }`

nameispriyank, 2020-09-08:
Stacktrace
`Fatal Exception: java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] j.a.a.c.b.a()' on a null object reference
       at me.aflak.bluetooth.Bluetooth$ReceiveThread.run(Bluetooth.java:556)`

Tho391, 2021-08-16:
Same issue:
`E/AndroidRuntime: FATAL EXCEPTION: Thread-15
    Process: com.bosch.aquaeasy.dev, PID: 30888
    java.lang.NullPointerException: Attempt to invoke virtual method 'byte[] j.a.a.c.b.a()' on a null object reference
        at j.a.a.a$i.run(SourceFile:556)`