omaraflak/Bluetooth-Library #60: Android Bluetooth Classic RFCOMM connection failing, getUUIDs() returns no SPP UUID

GraxCode, 2021-08-23:
I'm trying to send data using Bluetooth Classic between two paired Android devices.

One device is Huawei P Smart 2019 with EMUI Android 10, the other one a Pixel 4a running Calyx OS Android 11.

If I use the default 00001101-0000-1000-8000-00805f9b34fb UUID for a SPP / RFCOMM connection, it fails with read failed, socket might closed or timeout, read ret: -1 on both secure and insecure connection attempts. The paired device's BluetoothDevice#getUUIDs() only returns (on both ends)

0000110a-0000-1000-8000-00805f9b34fb
00001105-0000-1000-8000-00805f9b34fb
00001115-0000-1000-8000-00805f9b34fb
00001116-0000-1000-8000-00805f9b34fb
0000112f-0000-1000-8000-00805f9b34fb
00001112-0000-1000-8000-00805f9b34fb
0000111f-0000-1000-8000-00805f9b34fb
00001132-0000-1000-8000-00805f9b34fb
00000000-0000-1000-8000-00805f9b34fb
00000000-0000-1000-8000-00805f9b34fb

00001101-0000-1000-8000-00805f9b34fb (= SPP UUID) is nowhere to be found.

Some UUIDs (like 0000112f-0000-1000-8000-00805f9b34fb in the list work for connection (no error), however sending data does not work and gives no exception on both ends.

I also tried to use the reflection port workaround described in other issues, but no result (IOException: read failed, socket might closed - Bluetooth on Android 4.3)

Does one Android device simply not support SPP / RFCOMM or why does it not connect using the SPP UUID (1101)?

zaid-asif-brainx, 2022-01-13:
@GraxCode  did you find the solution?

loerting, 2022-01-16:
No, I didn't sadly.

tomeroto, 2024-03-06:
I have some problem too, how to solve this?